johnrbnsn/Adafruit_Python_MAX31856 #2: Bit shifting isn't correct

sjschaffer, 2017-01-09:
The bit shifting does not strip the unused bits from the low byte in the internal or external temp functions.

In readTempC(), this
        temp_bytes = ( ((val_high_byte & 0x7F) << 11) + (val_mid_byte << 3) + val_low_byte )
should be
        #put the bytes together
        temp_bytes = ( ((val_high_byte & 0x7F) << 16) + (val_mid_byte << 8) + val_low_byte)
        #Ignore the 5 least significant bits
        temp_bytes = temp_bytes >> 5
        
A similar correction is needed in readInternalTempC() to ignore the 2 least significant bits

BTW, thank you for creating this library. It accelerated my use of the MAX31856 on the Pi.

johnrbnsn, 2017-01-09:
I believe the unused bits end up as zeros (data sheet shows zeros for CJ, X's for Thermocouple temp), and since I'm scaling based on a known LSB I think our results should come out the same.  Your method is more explicit, mine *might* be quicker (one less operation).  I'll look into the data sheet to confirm, and then I'll update the code to your method (since the code is Python and readability will trump speed).  I'm traveling this week so it might not be til next that I update the code.

Thanks for using the library, and for commenting.  It's always good to know that the work gets used!

JR

johnrbnsn, 2017-02-24:
I finally had a chance to check this, turns out my understanding was wrong.  I've got some code below showing the differences between your (correct) suggestion, and my prior understanding.  I'm adjusting the code now to have the proper bit shifting, and then will close the issue.  Thanks for bringing this to my attention, 

JR

Python interpreter, Thermocouple Demo:
```
>>> # Dummy data from sensor
>>> byte2 = 0x01; byte1 = 0x70; byte0 = 0x20;  
>>>
>>> # Write the data in binary format for each bit shifting techinque, see clear difference in LSB
>>> format( ( (byte2 << 11) +(byte1 << 3)+ byte0   ), 'b' ).zfill(24)
'000000000000101110100000'
>>> format( ( (byte2 << 16) +(byte1 << 8)+ byte0   )>>5, 'b' ).zfill(24)
'000000000000101110000001'
>>>
>>> # See difference in measured temperature, small for this case, but could be up to 1.6953125 *C off, if LSB is 0xe0 (0b11100000), or the max value of the LSB.
>>> ((byte2 << 11) +(byte1 << 3)+ byte0)*2**-7
23.25
>>> ( ((byte2 << 16) +(byte1 << 8)+ byte0) >> 5)*2**-7
23.0078125
>>>
```

For Cold Junction temperature:
```
>>> MSB = 0x1C; LSB = 0x64;
>>> 
>>> format( ( (MSB << 6) + LSB   ), 'b' ).zfill(16)
'0000011101100100'
>>> format( ( (MSB << 8) + LSB ) >>2 , 'b' ).zfill(16)
'0000011100011001'
>>> ( (MSB << 6) + LSB )*2**-6
29.5625
>>> (( (MSB << 8) + LSB ) >> 2)*2**-6
28.390625
>>> # could end up with a max difference of 2.953125*C with a LSB of 0xfc, 0b11111100
```

closed 2017-02-24 by johnrbnsn