johnrbnsn/Adafruit_Python_MAX31856 #3: python3 support?

seanroark, 2017-01-17:
Followed the install instructions and I can import the library using python2 but not using python3. It is installed in python3.4/dist-packages/Adafruit_MAX31856.

Is the apt-get need something different for python3?

This is on a rasberry pi running latest Raspbian. 

[image]

johnrbnsn, 2017-01-17:
Did you install it with Python3 too?  If your pip is setup for Python2, then the module probably didn't install for Python3.  You can try to pip install using a 'venv' setup with Python3.  

Alternatively, copy the 'simpletest.py' into the same directory as 'Adafruit_Python_MAX31856' directory which contains 'max31856.py' and try running it directly.  

LMK, thanks.
JR

seanroark, 2017-01-17:
Its definitely pip installed in python 2 and 3. We are trying to get away from python 2 but a few things are keeping it around so we are well versed in the whole 2to3 thing.

johnrbnsn, 2017-02-22:
Took me a while to get back to this, I've updated the README to include new instructions which I've got working with Python3.  LKM what you think.  Still no GPIO support though, any progress on your end?  I'm closing the issue since I got it working in Py3.

closed 2017-02-22 by johnrbnsn