johnrbnsn/Adafruit_Python_MAX31856 #4: Is this only tested with SPI? Numerous problems with GPIO.

seanroark, 2017-01-17:
max31856.py line 112: The gpio variable does not exist. I'm not its intent but just initializing to None anywhere fixes it.

Also the GPIO import is named as Adafruit_GPIO but later referenced as GPIO on line 113.

_write_register only supports SPI.

Do you expect this to be GPIO compatible or is this SPI only? If so I'll do a pull request and try to get it working.

johnrbnsn, 2017-01-17:
I just got another email about this, I've only tested with Hardware SPI.  I was working off another example (Adafruits MAX31855 library) for the software SPI but never put any time into it.  Can software SPI do a direct XFER?  That's needed for the multi byte commands (AFAIK).

That would be great if you can get software SPI working, I'll gladly pull in any work you do (just be sure to write a test case).

Thanks for your help,
JR

johnrbnsn, 2017-06-16:
Hi Sean,

I think I've got the framework for GPIO in place in the latest commit https://github.com/johnrbnsn/Adafruit_Python_MAX31856/commit/436a672f145c7a5c8a5044a5d3694e067f9d278a .  I still don't use GPIO and I had some bugs pop up when I ran my tests (so I disabled GPIO tests), can you run this and see if it works for you?  I think the errors I got were from trying to use the same pins for both Hardware and GPIO SPI.

Thanks!
JR

As shown in /examples/simpletest.py :

    # Raspberry Pi software SPI configuration.
    software_spi = {"clk": 25, "cs": 8, "do": 9, "di": 10}
    sensor = MAX31856(software_spi=software_spi)

johnrbnsn, 2019-06-12:
No follow up, closing due to age. Please reopen or start a new issue if you still have the problem.

closed 2019-06-12 by johnrbnsn