johnrbnsn/Adafruit_Python_MAX31856 #5: Installation

bullvai, 2017-04-06:
Hey John. I feel like I'm missing something from your README. I see setup.py and ez_setup.py. Do I need to run these? What version of python does your code require? Python 3.4.1 doesn't seem to like:
$ python3 -m venv env_py3
Do I need pip? Why do you suggest to run your code in a virtual environment?

johnrbnsn, 2017-04-07:
You don't need to use the setup.py or ez_setup.py files directly, those are from before I converted things over to using pip.  (You can try `$ sudo python setup.py install` to see if that works for you).

An assumptions on my end: You're using Raspian (because you said Python 3.4).  I use Python 3.5 (or 3.6?); I don't remember exactly what forced me there, but there was an issue with Python 3.4 (maybe pip didn't install by default, and RasPi has both Python 2/3 versions installed, therefore pip/ pip3 issues?).  I've moved to the Ubuntu install on RasPi, because it was easier to do that than figure out how to install Python > 3.5 on Raspian.  

Using virtualenv is good practice, it keeps your dependencies isolated, and it makes it easy to transfer between machines.  You can $ pip freeze > requirements.txt on the machine your developing on, and $ pip install -r requirements.txt on another machine and end up with largely the same environment. Using the venv allows you to only install the dependencies you need, and not all the other crap you would probably have if you just installed packages to your main Python distribution.  With slow WiFi/ limited space on uControllers like the Pi it helps to only install what you need, and not other bloat.

Try the command `$ sudo python setup.py install` on your Raspian and let me know if that works.  I hope they update Raspian to Py 3.5+ soon (its 2017 and 3.5 has been out since late 2015!!)

JR

bullvai, 2017-04-08:
Thanks for your response JR. Here's what I've come up with to get it working in the virtual environment:

Here's my setup:
Python 3.4.1
Raspbian Jesse Lite (Kernal 4.4, release 03-02-17)

$ sudo apt-get update
$ sudo apt-get install -y  build-essential python-dev python-smbus python3-venv
$ sudo raspi-config  ---> Interfaces ---> Enable SPI

$ git clone https://github.com/johnrbnsn/Adafruit_Python_MAX31856.git
$ cd Adafruit_Python_MAX31856
$ python3 -m venv env_py3
$ source env_py3/bin/activate
(env_py3) $ pip install -r requirements.txt
(env_py3) $ pip install ./.

(env_py3) $ python simpletest.py was then successful.

Here's what I did to get it working in a normal environment:

$ sudo raspi-config -> enable SPI, change host name
$ sudo apt-get update
$ sudo apt-get install -y python-dev python-smbus python-pip python3-spidev
$ git clone https://github.com/johnrbnsn/Adafruit_Python_MAX31856.git
$ git clone https://github.com/adafruit/Adafruit_Python_GPIO.git
$ sudo mv /home/pi/Adafruit_Python_GPIO/Adafruit_GPIO /usr/lib/python3.4
$ sudo mv /home/pi/Adafruit_Python_MAX31856/Adafruit_MAX31856 /usr/lib/python3.4
$ /home/pi/Adafruit_Python_MAX31856/examples/python3 simpletest.py

In your max31856.py I added a few additional functions:

[image]

[image]

There was probably a more elegant way to do currentTcType() aside from a slew of elifs, but I'm a relative newbie so I ask for forgiveness!

Also I added type to the simpletest.py:

[image]

I can add these changes if you're interested, but I'm unfamiliar with how the process works. Thanks again for your excellent library by the way - you're awesome!

bullvai, 2017-04-08:
In my comment above I included a couple of unnecessary items:
Changing the host name is not required, it's just part of my larger project.
It is unnecessary to download python-pip.

johnrbnsn, 2017-04-10:
I'm glad you were able to get it working!  

Regarding your contributions, I think those are good ideas, but I'd like to keep the library as light as possible (also not much time right now to put into creating new tests for new methods).

JR

closed 2017-04-10 by johnrbnsn

bullvai, 2017-04-10:
Sounds good - thanks again!

As an aside, I ended up changing the thermocouple types to:

[image]

johnrbnsn, 2017-04-10:
Nice, that looks much more Pythonic!