johnrbnsn/Adafruit_Python_MAX31856 #6: Not able to install

anand2623, 2017-06-05:
python3 -m venv env_py3 
this does not execute and gives the following  error 

"Command '['/home/pi/Adafruit_Python_MAX31856/env_py3/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']' returned non-zero exit status 1"

Im not able to figure out what is the issue. Please help.

johnrbnsn, 2017-06-05:
What version of Python3 are you using?  Type python3 from the command line and paste the header lines, should look something like:

    john_rbnsn@instance:~$ python3
    Python 3.5.2 (default, Nov 17 2016, 17:05:23) 
    [GCC 5.4.0 20160609] on linux
    Type "help", "copyright", "credits" or "license" for more information.
    >>> 

From the error it looks like you don't have venv installed, I don't think it came standard on older versions of Python3.

anand2623, 2017-06-07:
Thanks john. I dint have the venv installed and i had version 3.4.2. 

I have another issue now, the cold end temperature is being read as 

Cold Junction Temperature -0.015625 deg. C
is this a conversion issue? I am pretty sure the temperature is not zero :)

closed 2017-06-07 by anand2623

johnrbnsn, 2017-06-08:
No Worries @anand2623 ! I appreciate you using the library, and the questions that go along with it. Were you able to get the negative temperature to work out too?

anand2623, 2017-06-12:
I work on high temperature ranges anything below 75 F is not necessary for me xD. But ill give the corresponding voltages and see if it works

anand2623, 2017-06-13:
Hey Jhon, I feel there is some issue with the write to CR1 register. Particularly this statement
self.CR1 = ((self.avgsel << 1) + self.tc_type)
It works for avgsel =0x0 (well adding zeroes), but for other options it fails. 

I feel it should be self.avgsel<<4. 
ex: avgsel =0x1 and tc_type = ox6
with existing code CR1 =  0x8
decimal = 000 1000
but with self.avgsel<<4 CR1 = 0x16 
decimal = 0001 0110
which sounds right as the avgsel is 6:4 and tc type is 3:0

Please let me know if I am thinking correctly about it.

johnrbnsn, 2017-06-13:
I opened a new issue and will respond here: https://github.com/johnrbnsn/Adafruit_Python_MAX31856/issues/7

Try to open new issues for separate problems, it helps with tracking.  Thanks!