johnrbnsn/Adafruit_Python_MAX31856 #7: CR1 Bit Shifting

johnrbnsn, 2017-06-13:
From @anand2623: 

Hey John, I feel there is some issue with the write to CR1 register. Particularly this statement
self.CR1 = ((self.avgsel << 1) + self.tc_type)
It works for avgsel =0x0 (well adding zeroes), but for other options it fails.

I feel it should be self.avgsel<<2.
ex: avgsel =0x1 and tc_type = 0x6 => CR1 is 0 001 0110
with existing code CR1 = 0xE (according to the log)
decimal = 0000 1110

but with self.avgsel<<2 CR1 = 0x16
decimal = 0001 0110
which sounds right as the avgsel is 6:4 and tc type is 3:0

Please let me know if I am thinking correctly about it.

johnrbnsn, 2017-06-13:
Good catch, but I think we actually want to shift by 4 (number of bits lower).  See example below:
    
    >>> self.avgsel = 0b001
    >>> self.tc_type = 0b0100
    >>> self.CR1 = ((self.avgsel << 4) + self.tc_type)
    >>> format(self.CR1, '#010b')
    '0b00010100'

I'm going to implement a fix, and check my other bit shifting parts to see how they work, along with implementing a new test to verify the fix.  Should be done tomorrow around lunchtime. 

Thanks again for finding the issue  @anand2623 !

anand2623, 2017-06-14:
Thanks John for fixing it. I thought it needs to shift by one more bit as it was giving 0000 1110 instead of 0001 0110. But you are right, shifting by 4 is the right way to do it. For now, I have bypassed this by writing the exact value to the CR1 register directly.

johnrbnsn, 2017-06-16:
@anand2623 , commit https://github.com/johnrbnsn/Adafruit_Python_MAX31856/commit/436a672f145c7a5c8a5044a5d3694e067f9d278a should solve your bitshifting problem, along with a few function renames to meet PEP8 (I've been meaning to make those changes).  This commit still contains the old function names (old: readTempC -> new: read_temp_c) as deprecated methods, so you can still use the old names, but try to move to the new calls if you can. In the future the old method names might be removed.

Please test and let me know what you get.

closed 2017-06-16 by johnrbnsn