johnrbnsn/Adafruit_Python_MAX31856 #8: No module named Adafruit_MAX31856

ayebayley, 2017-10-09:
I having some difficulties getting up and running. I had everything working just fine on a different Pi (with software SPI tested up to 4 MAX31856 at a time) but now on a new installation, I can't get anything rolling.

I followed the instructions exactly, and I'm getting an ImportError: No module names Adafruit_MAX31856.

Also, how do I run this outside of the virtual environment? I would like to include this in another script but I'm not sure how to do that with the startup requirements here.

johnrbnsn, 2017-10-23:
Hey, Sorry for the delay. 

It looks like Python can't find the code, which might happen if you had the virtualenv activated when you installed the library, and don't have the library activated when running the code. From the directory where you have the 'env_py3' folder, did you run 'source env_py3/bin/activate' ? Then try to run the script/ install the library.

Run outside virtualenv: This ones pretty easy, just skip the steps for creating and activating the virtualenv and run the rest as normal.  No real need to do this though, you can install the other script in the same virtualenv, or just activate the venv and run the other script. venv's help keep things clean, and avoid problems with different dependencies.

johnrbnsn, 2019-06-12:
Closing this issue, user install problem, no follow up.

closed 2019-06-12 by johnrbnsn