johnrbnsn/Adafruit_Python_MAX31856 #9: negative reading

petermrtn, 2017-10-23:
Hi,

I have tried unsuccessfully with your code to get the temperature reading of my thermocouple K with my Raspberry Pi 3.  The simpletest code returns always the following two values: -0.00756 and -0.00354. 

I have wired the MAX31856 as described in Adafruit website, I have checked multiple times the connections and I have checked with a multi-meter that the thermocouple returns normal readings. Any help would be greatly appreciated. I am stuck and I don't know what else I could do

Thank you very much in advance

johnrbnsn, 2017-10-23:
Did you use the tc_type argument when initializing the sensor?

    sensor = MAX31856(tc_type=MAX31856.MAX31856_K_TYPE, hardware_spi=Adafruit_GPIO.SPI.SpiDev(SPI_PORT, SPI_DEVICE))

This needs to be included if you're using a sensor other than the default (T_TYPE) in order to use the right scale for converting the thermocouple voltage to a temperature.

petermrtn, 2017-10-23:
Hi,
thank you for the prompt answer.
Yes, I did. I did changed the defaults at max31856.py. But I have even tried with the following to initialize the sensor:
software_spi = {"clk": 25, "cs": 8, "do": 9, "di": 10}
sensor = MAX31856(tc_type=MAX31856.MAX31856_K_TYPE, software_spi=software_spi)

Unfortunately I still get temp=-0.008*C and Internal Temperature: -0.016*C

Thanks

charles-d-burton, 2018-09-29:
I'm having the same exact issue, even the numbers are the same.

johnrbnsn, 2018-09-30:
I'll try to look into this next week, work has taken me in a different direction so no guarantee I'll be able to get to it. If you find a solution, lmk and I can integrate it.

maxi1008, 2019-04-12:
Hi!
I had the same issue, also with the same numbers and the problem was a bad connection of the sensor ground pin.

johnrbnsn, 2019-06-12:
k-type appears to be working now, see [this issue](https://github.com/johnrbnsn/Adafruit_Python_MAX31856/issues/12) for resolution and example. If you're having a different problem, please reopen issue or create a new one.

closed 2019-06-12 by johnrbnsn