johnrbnsn/Adafruit_Python_MAX31856 #12: Temperature reading stops at a certain value

PauloFisch, 2019-04-18:
Hey!

I am trying to use the Max31856 board from Adafruit to read temperatures of around 550C with a K Type thermocouple. However, whenever the thermocouple temperature goes above 399.984C it simply doesn’t increase. All readings afterwards present this value.
Does this have to do with the Bytes assigned to the reading or any setup on the code? I’ve tested multiple thermocouples, all with the same result.

Thanks!

johnrbnsn, 2019-04-18:
Hi,

If you're using Adafruits K-Type (https://www.adafruit.com/product/270), it looks like its only good to 500°C, so you won't be able to get to 550°C with that type anyway.

I'm looking at the code now to see if there is a software issue with the limit of 500°C. 

Do you know how to check registers on the component? If you can check to see if the chip is hitting the high temperature fault? TC High FAULT register will be triggered. It's been a while since I've used this code, so I'd have to dig and see if I built a method for register checking.

I'll dig and let you know what I find....

PauloFisch, 2019-04-18:
Hi! Thanks for the quick reply!

I am using Omega Type K thermocouples. These are rated to temperatures over 1000C.
Also, I have not used the fault register, as I was not so sure on which pin it should be connected. The issue seems to me at first glance like an overflow problem, as the readings suddenly stop at a very definite value.

Could it be somehow connected to the fact I am using Software SPI?

Once again, thank you so much for the help!

johnrbnsn, 2019-04-18:
Good to know on the thermocouples. Have you run the test suite yet? If not, can you run `python test_MAX31856.py -v` as shown in the readme and paste the output here.

As far as Software SPI, I know I had trouble with either Software or Hardware SPI but it's been so long I don't remember which. In my tests software spi test is commented out, so it was probably software I was having trouble with. Can you run it with Hardware SPI?

PauloFisch, 2019-04-18:
I have run the test suite recently. Unfortunatey I don’t have my assembly with me righ now though, only in a few days I’ll work more on it. The only errors I recall are regarding no Hardware SPI detected when using Software SPI (which seems reasonable to happen, I guess).

I have also used Hardware SPI before. It worked then, but now when I tried to change it again, there was no reading.

Also, I’m considering trying some other types of thermocouples, such as the Type B. Perhaps the voltage generated by the type K thermocouple is way too high for the breakout.

johnrbnsn, 2019-04-18:
I'm guessing you're getting the no Hardware SPI because it's not activated on your board. You said you've used it before, but perhaps the OS was wiped/ changed? Try [instructions from readme](https://github.com/johnrbnsn/Adafruit_Python_MAX31856#make-sure-spi-is-enabled-on-your-pi) to confirm if it is on or not.

Yeah, could be, but it sounds more like something digital to me. With analog output, I doubt it would be as repeatable as you describe. 

LMK the result of the test suite, and the Hardware SPI check.

PauloFisch, 2019-05-03:
Hey!

I`ve tried the test suite again and here is the result:

E
Warning (from warnings module):
  File "/home/pi/Adafruit_Python_MAX31856/Adafruit_MAX31856/test_MAX31856.py", line 37
    GPIO.cleanup()
RuntimeWarning: No channels have been set up yet - nothing to clean up!  Try cleaning up at the end of your program instead!
EEE.
======================================================================
ERROR: test_get_internal_temperaure_reading (__main__.Adafruit_MAX31856)
Checks to see if we can read a temperature from the board, using Hardware SPI
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/pi/Adafruit_Python_MAX31856/Adafruit_MAX31856/test_MAX31856.py", line 132, in test_get_internal_temperaure_reading
    sensor = MAX31856(hardware_spi=SPI.SpiDev(spi_port, spi_device))
  File "/home/pi/Adafruit_Python_MAX31856/Adafruit_MAX31856/max31856.py", line 111, in __init__
    self.tc_type = MAX31856_K_TYPE
NameError: global name 'MAX31856_K_TYPE' is not defined

======================================================================
ERROR: test_get_register_reading (__main__.Adafruit_MAX31856)
Checks to see if we can read a register from the device.  Good test for correct
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/pi/Adafruit_Python_MAX31856/Adafruit_MAX31856/test_MAX31856.py", line 82, in test_get_register_reading
    sensor = MAX31856(hardware_spi=SPI.SpiDev(spi_port, spi_device))
  File "/home/pi/Adafruit_Python_MAX31856/Adafruit_MAX31856/max31856.py", line 111, in __init__
    self.tc_type = MAX31856_K_TYPE
NameError: global name 'MAX31856_K_TYPE' is not defined

======================================================================
ERROR: test_get_temperaure_reading (__main__.Adafruit_MAX31856)
Checks to see if we can read a temperature from the board, using Hardware SPI
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/pi/Adafruit_Python_MAX31856/Adafruit_MAX31856/test_MAX31856.py", line 116, in test_get_temperaure_reading
    sensor = MAX31856(hardware_spi=SPI.SpiDev(spi_port, spi_device))
  File "/home/pi/Adafruit_Python_MAX31856/Adafruit_MAX31856/max31856.py", line 111, in __init__
    self.tc_type = MAX31856_K_TYPE
NameError: global name 'MAX31856_K_TYPE' is not defined

======================================================================
ERROR: test_software_spi_initialize (__main__.Adafruit_MAX31856)
Checks to see if the sensor can initialize on the software SPI interface.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/pi/Adafruit_Python_MAX31856/Adafruit_MAX31856/test_MAX31856.py", line 49, in test_software_spi_initialize
    sensor = MAX31856(software_spi=software_spi)
  File "/home/pi/Adafruit_Python_MAX31856/Adafruit_MAX31856/max31856.py", line 111, in __init__
    self.tc_type = MAX31856_K_TYPE
NameError: global name 'MAX31856_K_TYPE' is not defined

----------------------------------------------------------------------
Ran 5 tests in 0.047s

FAILED (errors=4)

It seems now to have no issue whatsoever with hardware SPI tho. I just find it weird that it is not recognising the K Type as a variable, as it is announced at the max31856 module. Can it be that it is using the T Type instead because it is default?

johnrbnsn, 2019-05-07:
Sorry for the delay. What python version are you using?

PauloFisch, 2019-05-08:
I’m using Python 2.7. I have tried 3.4 prior to that, but the library import didn’t seem to work.

PauloFisch, 2019-06-04:
Hey, just leaving an update:

I finally found out why the temperatures aren`t reading past 400C. It is because the module is not being reloaded correctly after the change to type K thermocouple has been made. Hence, it is still measuring everything based on a type T thermocouple, which has a maximum temperature of 400C and is the default of the library.

I`ve tried using the reload(library) function on Python 2.7, but it doesn`t seem to recognise the module, any suggestions?

Cheers,

johnrbnsn, 2019-06-10:
Can you paste your code where you initialize and read the temperature? A minimal example. I think it should be:

    from max31856 import MAX31856 as MAX31856
    temp_sensor = MAX31856(tc_type=MAX31856.MAX31856_K_TYPE)
    temp_sensor.read_temp_c()

Can you try that and let me know if that works?

PauloFisch, 2019-06-12:
I`ve tried using this one you sent and it did work. The issue was, I created the object "sensor" without announcing the thermocouple type, as such:

sensor1 = MAX31856(software_spi=software_spi1)

now, using:
sensor1 = MAX31856(software_spi=software_spi1,tc_type=MAX31856.MAX31856_K_TYPE)

it does now in fact work.

Thank you so much for the support, finally I can go on with my project :)

johnrbnsn, 2019-06-12:
Great, thanks for sticking with the code Paulo! I'll add an example to the tests and examples today to help others in the future.

closed 2019-06-12 by johnrbnsn