johnrbnsn/Adafruit_Python_MAX31856 #14: test_MAX31856.py --> FAIL

eXact07, 2019-08-30:
I have a problem running the test file "test_MAX31856.py".  I use a Raspberry Pi 3 B+ and installed everything like it's described in the instruction from "johnrbnsn". The test files "simpletest.py" and "simpletest_k_type.py" are working like it's described. But when I run the file "test_MAX31856.py" it fails and there appears the following code lines:

`test_get_internal_temperaure_reading_k_type (__main__.Adafruit_MAX31856) ... test_MAX31856.py:42: RuntimeWarning: No channels have been set up yet - nothing to clean up! Try cleaning up at the end of your programm instead!
    GPIO.cleanup()
FAIL
test_get_internal_temperaure_reading_k_type (__main__.Adafruit_MAX31856) ... FAIL
test_get_register_reading (__main__.Adafruit_MAX31856) ... FAIL
test_get_temperaure_reading (__main__.Adafruit_MAX31856) ... FAIL
test_hardware_spi_initialize (__main__.Adafruit_MAX31856) ... ok
test_temperature_byte_conversions (__main__.Adafruit_MAX31856) ... ok

===============================================
FAIL: test_get_internal_temperaure_reading (__main__.Adafruit_MAX31856)
---------------------------------------------------------------------------------
Traceback (most recent call last):
    File "test_MAX31856.py", line 148, in test_get_internal_temperature_reading
        self.assertTrue(False)
AssertionError: False is not true

===============================================
FAIL: test_get_internal_temperaure_reading_k_type (__main__.Adafruit_MAX31856)
---------------------------------------------------------------------------------
Traceback (most recent call last):
    File "test_MAX31856.py", line 165, in test_get_internal_temperature_reading_k_type
        self.assertTrue(False)
AssertionError: False is not true

===============================================
FAIL: test_get_register_reading (__main__.Adafruit_MAX31856)
---------------------------------------------------------------------------------
Traceback (most recent call last):
    File "test_MAX31856.py", line 99, in test_get_register_reading
        self.assertTrue(False)
AssertionError: False is not true

===============================================
FAIL: test_get_temperaure_reading (__main__.Adafruit_MAX31856)
---------------------------------------------------------------------------------
Traceback (most recent call last):
    File "test_MAX31856.py", line 131, in test_get_temperature_reading
        self.assertTrue(False)
AssertionError: False is not true

---------------------------------------------------------------------------------
Ran 6 tests in 0.029s

FAILED (failures=4)`

Can somebody help me? I don't know how to fix this problem...

Thank you very much for your help :)

closed 2019-09-04 by eXact07

johnrbnsn, 2019-09-04:
Please share your solution in case anyone else has the same problem. 

Thanks!

eXact07, 2019-09-06:
I just wired the MAX31856 boards wrong, because I used the "board-numbers" and not the "bcm/gpio-numbers".