johnrbnsn/Adafruit_Python_MAX31856 #17: Problem setting up

JeremyRiggi, 2020-04-22:
Hello !
,
I wish to carry out temperature measurements with the MAX31856 module.
However, I encounter two problems: 
After following the instructions I tried to run the program 'simpletest_k_type".
However he says "No module named 'Adafruit_MAX31856'. I think problem comes from the __init__.py file when I run it says "ModuleNotFoundError : No module named '__main__.max31856' ; '__main__' is not a package

Second concern about the game (env_py3) $ python simpletest.py 
When I do the tests, it tells me [Errno 2] No such file directory

Thank you for your help

whomble, 2023-06-16:
Hi jeremy, 

I have the same issue did you found a solution?

JeremyRiggi, 2023-06-16:
Hi ! It's been a while now and I don't remember ^^ but  except that the big problem was that I hadn't soldered the sensor and the contact was only half made.

whomble, 2023-06-16:
Thanks for the fast reply ^^
I think the issue is more about the installation of the library or something like taht because it happend even when the sensor isn't connected
I recieve this everytime no matter which pin i used
"ModuleNotFoundError : No module named 'main.max31856' ; 'main' is not a package

Second concern about the game (env_py3) $ python simpletest.py
When I do the tests, it tells me [Errno 2] No such file directory

whomble, 2023-06-20:
Okay it works. 
I've removed "simpletest.py" from the exemple folder and paste it directly in the folder "/home/user/Adafruit_Python_MAX31856" 
I've also try using software spi instead of hardware spi and it solve all my problems!
Thanks

johnrbnsn, 2023-06-20:
@whomble Glad you got it running.

closed 2023-06-20 by johnrbnsn